ut-issl/c2a-aobc #133: ExampleでのvMicroビルド設定の修正

200km, 2023-07-20:
## 詳細
ISSL内プロジェクトで使っている`c2a-aobc-衛星名`のレポジトリで、vMicroビルド設定を追加する必要があるとわかったので、Exampleも同様に修正する。

## close条件
ビルド設定が更新されたら

## 備考
NA

## 注意
NA

200km, 2023-07-20:
@seki-hiro mainからhotfixで修正してもらえると嬉しいです。パッチバージョンアップデートもやってみてください。

200km, 2023-07-20:
> パッチバージョンアップデートというのは、タグv6.0.2を切る

そうです！
- mainを修正
- 修正後のmainに対してタグv6.0.2を切り、そのtagを使ってv6.0.2をリリース
- main -> develop PRを投げてdevelopで修正を取り込む

という流れなので一度やってみてください。

seki-hiro, 2023-07-21:
@200km ディレクトリ構成で混乱してきたので、確認させてください。

[How to make a project-specific repository](https://github.com/ut-issl/c2a-aobc#how-to-make-a-project-specific-repository)にあるように、`src-aobc (git submodule of c2a-aobc with rename)`と`Build setting files`は同じ階層にあることを想定していると思います。したがって、ユーザー向けに例を示す場合は、c2a-aobcとは別レポジトリでc2a-aobc-exampleを公開するのが良いのではないでしょうか？

Examples/CMakeLists.txt内には、src_aobcという記述がありますが、現在のc2a-aobcにはそのようなフォルダはなく、vMicroビルドだけでなく、CMakeの構成に失敗します。過去のディレクトリ構成が残っているのだと思いますが、Example周りを最新のディレクトリ構成の想定にアップデートしたいです。

200km, 2023-07-21:
[RUN_AFTER_CLONE](https://github.com/ut-issl/c2a-aobc/blob/develop/RUN_AFTER_CLONE.bat)は実行していますか？これを実行すると、Example/src_aobcができて、そこにc2a-aobcがコピーされます。この状態でExample以下がテスト用レポジトリのように見えるというのが今の想定です。

> c2a-aobc-exampleを公開するのが良いのではないでしょうか？

私はこれが良いと思っているのですが、以前の議論で`Exampleディレクトリを作る`ということになり、このようなスタイルになっています。  
@suzuki-toshihir0 コメントいただけると嬉しいです。

seki-hiro, 2023-07-21:
すみません。環境構築が足りていなかっただけかもしれません。確認してみます。

suzuki-toshihir0, 2023-07-21:
Example dirを作るのは c2a-coreと同様の発想です： https://github.com/ut-issl/c2a-core/tree/develop/Examples
実際に実機に書き込まれうるリポジトリであれば別リポジトリに切り出して公開するというのはわかるのですが，Exampleに関してはそうではなく，単にサンプルが示せればよいので同一リポジトリで良いという考えです．

200km, 2023-07-21:
ユーザー目線で見ると、`c2a-aobc-example`があり、Actionsなども含めてコピーできたら楽ということはある気がしています。  
s2e/c2aでも [s2e-user-for-c2a-core](https://github.com/ut-issl/s2e-user-for-c2a-core)というサンプルレポジトリがあったりもします。

200km, 2023-07-21:
> 環境構築が足りていなかった

READMEが分かりづらいのが問題かもなので、必要なら追記してください。

seki-hiro, 2023-07-24:
.batファイルを実行したら問題なく修正できました。慣れでREADME読まずに進めてしまったのが原因で、わかりづらくはないです。

seki-hiro, 2023-07-24:
Exampleを別レポジトリにするのは一長一短あるということが理解できました。今後ユーザーが増える中で、あまりにも質問が多いとかであれば、

> c2a-aobc-exampleがあり、Actionsなども含めてコピーできたら楽

を整備することになりそうですね。

closed 2023-07-24 by seki-hiro