ut-issl/c2a-aobc #160: Sagittano

seki-hiro, 2023-08-04:
## 詳細
詳しく

## close条件
どうなったらcloseできるか

## 備考
なにかあれば

seki-hiro, 2023-08-24:
メーカー側で確認中のため、継続対応します。

seki-hiro, 2023-09-01:
EM固有の問題とのこと。FMで試験し、問題なければcloseする。

conjikidow, 2024-02-26:
FMで試験した結果，EMのように `-1` とはならなかったが，全て `0` となった。
[image]

conjikidow, 2024-02-26:
FMで試験した結果，問題なくテレメが得られた。
[image]

@seki-hiro 一応close前に確認お願いできますか？

seki-hiro, 2024-03-07:
確認ありがとう！

CMVPixだけ、他と比較して電流と電圧が入れ替わってそうですが、テレメ周りのコード見ても特段問題なさそうなので、そういうものっぽいですね。

-1以外の値が入ることは確認できたのと、1.8, 2.5, 2.1, 3.3V出るべきラインは出ているので、closeで問題ないと思います。

conjikidow, 2024-03-08:
確認ありがとうございます。

closed 2024-03-08 by conjikidow