ut-issl/c2a-aobc #271: 目標計算に対地表面相対速度を入れる

200km, 2024-02-08:
## 詳細
目標計算時に今は慣性座標系での速度方向を指定する機能が入っているが、対地表面相対速度を指定できるように機能を追加する。

## close条件
追加したら。

## 備考
NA

Hiro-0110, 2024-02-18:
メモ
[ここ](https://github.com/ut-issl/c2a-aobc/blob/d44547e787de14bc2171001133fe3691ee795384/src/src_user/Applications/UserDefined/AOCS/TargetAttitude/target_attitude_from_orbit.c#L184) に追加する

Hiro-0110, 2024-02-18:
メモ
[s2e](https://github.com/ut-issl/s2e-core-for-tdi-test/pull/10/files#diff-53dc1135aeb731a2a405951a055d129e2c9a938a81d89bae27c49ed5283c7006)側の実装がこれ

Hiro-0110, 2024-02-18:
メモ
ecef座標系での速度を[ここ](https://github.com/ut-issl/c2a-aobc/blob/d44547e787de14bc2171001133fe3691ee795384/src/src_user/Applications/UserDefined/AOCS/aocs_manager.c#L528)からとってくる

Hiro-0110, 2024-02-19:
ecef to eci https://github.com/ut-issl/c2a-aobc/blob/d44547e787de14bc2171001133fe3691ee795384/src/src_user/Applications/UserDefined/AOCS/TargetAttitude/target_attitude_from_orbit.c#L218

closed 2024-04-17 by Hiro-0110