ut-issl/c2a-aobc #290: BC数の最大値を増やすことを検討する

conjikidow, 2024-05-08:
## 詳細
これまでの運用実績から，軌道上で追加することも考えるとBC数の最大値を大きくしたほうが良いと考えられる。
メモリ余裕を評価しながら，増やすべきか，増やすとしたらどれほど増やせるかを検討する。

## close条件

- [ ] BC数上限を増やすか決まったら
- [ ] 増やす場合，いくつ増やすか決まったら

## 備考
NA

Hiro-0110, 2024-05-08:
block_command_table_params.hに記入する

Hiro-0110, 2024-05-08:
BCT_MAX_BLOCKSを増やして行った時のメモリ使用量の増加はこんな感じです

[image]

Hiro-0110, 2024-05-09:
@200km @conjikidow @seki-hiro 
メンションしておきます

Hiro-0110, 2024-05-10:
@200km @conjikidow @seki-hiro 
c2a-aobc-onglaisatでの結果です
[image]

Hiro-0110, 2024-05-10:
@200km @conjikidow @seki-hiro 
結果です
[image]

conjikidow, 2024-05-10:
ありがとうございます。10個増やして90個にして良い気がしますね。

Hiro-0110, 2024-05-10:
https://github.com/ut-issl/c2a-aobc/pull/295 プルリク出しました

Hiro-0110, 2024-05-10:
@conjikidow 
https://github.com/ut-issl/6u-aocs-task/issues/74
このAOBC BCを追加する作業もした方がいいですか？

conjikidow, 2024-05-10:
とりあえず追加しないで大丈夫です。