ut-issl/c2a-aobc #304: devでvMicro buildができない

seki-hiro, 2024-05-17:
## 詳細
#282 のdevマージからvMicroでbuildできなくなってしまっている

## close条件
vMicroでbuildできるようになったら

## 備考
なにかあれば

200km, 2024-05-17:
#282　で追加されたベクトル変数がヘッダでしか定義されておらず、値が入れられていないことが原因かもしれない。
例えば、`OEM7600_PARAMETERS_mag_bias_compo`に値を入れるコードを`Sample/oem7600_parameters.c`の方にも書かないと、c2a-aobc単体で動かした時に値が定義されない。

SILSやchipkit toolchainなどの場合は、初期化されていないというだけでビルドできるかもしれないが、vMicroではエラーが出ているのかもしれない。

ただ、これが原因でなかったとしてもあたいは入れるべきなのでどのみち修正は必要。

200km, 2024-05-17:
あと、ついでに命名規則について次の点を直してもらいたいです。
- 単位を最後につけてください
- AOBC、EXTなどは小文字が良いです

`NANOSSOC_D60_PARAMETERS_mag_bias_compo_AOBC`
は
`NANOSSOC_D60_PARAMETERS_mag_bias_aobc_compo_nT`
が良いと思います。

本当はMPUではなく、RM3100に対するバイアスであることがわかるようにしたいので、
`NANOSSOC_D60_PARAMETERS_mag_bias_ｒｍ３１００_aobc_compo_nT`
や
`OEM7600_PARAMETERS_mag_bias_rm3100_ext_compo_nT`
(以下同様)
が良いかもしれません。

closed 2024-05-18 by ogoogo