ut-issl/c2a-aobc #312: BdotモードでMTQがまったく出力されない場合がある

suzuki-toshihir0, 2024-05-21:
## 詳細

Bdotモードのタスクリストの組み方・磁気排他制御タイマーの状態によってはMTQが全く出力されないケースがあることが判明した。

### 前提

Bdotモードの現状のタスクリストは以下である：

https://github.com/ut-issl/c2a-aobc/blob/c8caecd3ebc82ceb39e8fe8c3b62600635d2a347/src/src_user/Settings/Modes/TaskLists/tl_bdot.c#L13-L36

### 現状の仕様と実装

磁気制御がタスクリストに沿ってどのように実行されるかを理解するためには、各アプリの仕様と実装を正しく理解しておく必要がある。

- 磁気排他制御タイマー：stateが `OBSERVE` （磁気センサを使った磁場観測とそれに伴う磁場推定・出力目標値計算に使って良い時間枠）-->  `CONTROL` （計算した出力目標値を使ってMTQを出力するのに使って良い時間） --> `STANDBY` （消磁待ちに使って良い時間）と順番に切り替わる。磁気排他制御タイマーは他のオブジェクトのGetter / Setterを叩いて操作することはせず、他のアプリが磁気排他制御タイマーのstateを見ることによって排他制御を実現する。 
- 磁気センサDI：DIが呼ばれるたびに、磁気計測値が磁気センサドライバ構造体のメンバに保存される。
- 磁気センサフィルタ：磁気排他制御タイマーが `OBSERVE` のときに限り、スパイクフィルタ（とLPF）を実行してAOCS managerの磁場ベクトル推定値に値をpushする。
- Bdotアプリ：AOCS managerの磁場ベクトル推定値の直近2回の値を自身のメンバに保持する。その2回の値から磁場ベクトルを微分し、AOCS managerの目標出力磁気モーメントに値をpushする。
- MTQコントローラー：磁気排他制御タイマーのstateが `OBSERVE` のとき、AOCS managerの目標出力磁気モーメントを読み、3個のMTQに対してそれぞれが何秒間出力すればその磁気モーメントを実現できるかを計算する。続いて、stateが `CONTROL` のとき、MTQ DIに磁気モーメント出力指令をする。
- MTQ DI：ただのGPIO制御のラッパーであり、MTQコントローラーから指示されたらGPIOポートから電流を流す。

### MTQが出力されるケース

ノミナルケースである磁気センサにRM3100を使う場合は、それぞれの状態量が時間に沿って以下のように変化していくため、MTQが正しく出力される。

[image]

### MTQが出力されないケース

RM3100を使っているがstate timerのアップデートタイミングが50msズレている場合・MPU9250を使っている場合は、それぞれの状態量が時間に沿って以下のように変化していくため、MTQが出力されなくなってしまう。

- RM3100を使っているが、step == 40のときに `magnetic_exclusive_control_timer->state_timer_ms` が50となるケース

[image]

- MPU9250を使っていて、step == 40のときに `magnetic_exclusive_control_timer->state_timer_ms` が0となるケース

[image]

- MPU9250を使っていて、step == 40のときに `magnetic_exclusive_control_timer->state_timer_ms` が50となるケース

[image]

## close条件
どうなったらcloseできるか

## 備考
なにかあれば

suzuki-toshihir0, 2024-05-21:
@seki-hiro いえ、タスクリストの変更がないように対処します。具体的な対処は次の2点になる予定です。

1. 以下のobserve_durationを200に設定する

https://github.com/ut-issl/c2a-aobc/blob/c8caecd3ebc82ceb39e8fe8c3b62600635d2a347/src/src_user/Applications/UserDefined/AOCS/ExclusiveControl/magnetic_exclusive_control_timer.c#L33-L35

2. （これまでの衛星と制御系の挙動をあわせるため）以下を1000msに設定する

https://github.com/ut-issl/c2a-aobc/blob/c8caecd3ebc82ceb39e8fe8c3b62600635d2a347/src/src_user/Settings/SatelliteParameters/Sample/attitude_control_parameters.c#L11

seki-hiro, 2024-05-21:
干渉しないということでよかったです。ありがとうございます。

conjikidow, 2024-05-27:
#313 で対応したのでcloseする

closed 2024-05-27 by conjikidow